dojoe/Twonkie #1: Purpose of discontinuous CC2?

henryrgithub, 2021-03-23:
I'm confused as to the purpose of having a discontinuous CC2 on the board. Do you have to ensure correct cable orientation between Twonkie, the host and the device? On first blush it looks like twisting the cable could leave no CC line between the device and the host.

dojoe, 2021-03-23:
CC2 isn't discontinuous, it's just taken through a voltage/current monitor device; the INA260 has an internal shunt resistor for monitoring current so the signal goes right through it.

Depending on plug orientation either CC1 or CC2 is used for the actual control channel, and the other line becomes VCONN, the supply voltage for active circuitry in the plugs and/or cable. With the INA260 on one of the CC lines Twonkie is able to monitor VCONN voltage and current in one of the two orientations. Control channel communication will work in either orientation since both lines are continuous (modulo a very low value shunt resistor).

Does this help clear up your confusion? :)

henryrgithub, 2021-03-23:
Ah yes, that makes sense! Thanks for the clarification.

closed 2021-03-23 by henryrgithub