dojoe/Twonkie #4: EPR support

dnschneid, 2021-05-26:
Unfortunately, the INA260 only goes up to 36V (and will pop above 40V).  The new EPR standard goes up to 48V.

I don't see any TI INAs that do the higher voltage with integrated sense resistors, unfortunately.

dojoe, 2021-05-28:
Dang, too bad. Looks like I should replace the INA260 by an INA237 and an external shunt then. There goes my 1"x1" envelope 😭 
Before I go and rip the layout apart, can you think of anything else that would cause problems with recent PD developments - such as the F072CB eventually being unable to host a firmware capable of decoding the byzantine PD protocols?

dnschneid, 2021-05-28:
The F072CB is fine if you're using sigrok/pulseview to decode stuff, since it's just passing along the bits.

dojoe, 2023-01-29:
Well, that took a while -.-
I've got a beta v2.0 layout now that sports the INA237 and still fits the 1in by 1in envelope :)
Would you be willing to take a glance at 59ddbe225d1126198d7e3ff15a19b13e037b3664 and point out any glaring mistakes? The bottom half of the board hasn't changed but the top half has. I'm mostly worried about the current carrying capabilities of the VBUS copper planes, and perhaps the odd trace/via snuggling up too closely to the high speed traces.

dnschneid, 2023-02-02:
The layout looks pretty good!  VBUS current capacity seems totally fine; you have two planes (effectively) and plenty of vias per pin.

Just comments/nits here:

 * You have the sense resistors set up for diagonal conduction.  I assume this is to make layout a little easier, but the datasheet recommends conducting on one side and sensing on the other.  I have no idea how the device is constructed so my best guess is that accuracy could suffer.  Honestly the part is a little weird -- I'd have expected a dedicated kelvin sense resistor to be asymmetric to reduce the conduction losses at the solder joint, but maybe the part is just too small to do that and still be easily assembled.
 * I'm also a little surprised you swapped out the Vconn sense as well; the old one would have worked just fine.  I suppose it makes the firmware easier though, as does keeping the same sense resistor value.  I don't remember what we did on Twinkie since the chromium.org site seems to be borked and not showing attachments anymore.
 * You could make VRCONN+ hug the +3V3 trace on inner2 and let vbus or gnd flood the remainder.
 * You may also want to connect the VBUS_RECEP shape on inner2 connect with the vias just north of them.
 * 48V is the point at which you may want to consider spacing requirements.  IPC-2221B says 4 mils (0.1mm) on internal layers and 24 mils (0.6mm) on external layers.  I'm not sure you really need to follow that spec for a testing device that's not on all the time, so if those constraints mess up your layout I wouldn't bother.
 * Nice job squeezing the D+/D- between the sense resistor leads :wink:  Maybe shove the left half down a smidge to center the pair between CC1 and SBU1.
 * RX2 is definitely cramped.  Not much I can recommend that wouldn't result in a big rip-up,  as far as I can tell.  The proximity of the SBU1 trace is the worst offender, so if there's anywhere else you can route that, that may be worthwhile.  I can see adding a via on the right side to route it next to SBU2, but after that it gets pretty hairy.  This is an impressively dense layout!

dojoe, 2023-02-04:
Thanks for the thorough review! Applying your suggestions turned out to be easier than expected :)
I updated the `epr` branch with an updated layout.

> You have the sense resistors set up for diagonal conduction.  I assume this is to make layout a little easier, but the datasheet recommends conducting on one side and sensing on the other. 

I blame that one on the KiCad footprint library ;) Well spotted, thank you!
Turns out that this is relatively easy to fix if I rotate the VBUS sense resistor by 90 degrees and the result looks a lot better :)

>  I'm also a little surprised you swapped out the Vconn sense as well; the old one would have worked just fine.

I don't remember my rationale for that but I guess part of it was BOM reduction - why depend on two different pieces of Unobtainium when you could depend on one instead? ;)

>  You could make VRCONN+ hug the +3V3 trace on inner2 and let vbus or gnd flood the remainder.

The goal being less board area to etch, correct? Makes sense to me - I generally tried to grow all pours where easily possible now.

>  You may also want to connect the VBUS_RECEP shape on inner2 connect with the vias just north of them.

My thinking was to separate the current flow originating from the top and bottom pair of VBUS pins to evenly spread the load between the planes, I guess I was overthinking things there and should just let the electrons do their thing? :)

>  48V is the point at which you may want to consider spacing requirements.  IPC-2221B says 4 mils (0.1mm) on internal layers and 24 mils (0.6mm) on external layers.

Wouldn't the Twonkie fall more into class B4 due to the solder mask though, making the clearance requirement for outer layers 5 mils / 0.13mm?
I'm currently at 6 mils in my DRC setup so B4 would be satisfied. I'm not sure I could even escape the Type-C connector at 24mil spacing 😅 

>  Nice job squeezing the D+/D- between the sense resistor leads 😉  Maybe shove the left half down a smidge to center the pair between CC1 and SBU1.

Yup, and it looks nicer too :)

>  RX2 is definitely cramped.

There was still some space above the sense resistor to move it up a smidge, so I could move SBU1 away from RX2. Now VCONN_ALERT_L and that 3V3 via are the worst offenders but those aren't going to budge :) 

> This is an impressively dense layout!

Thank you 😊

dnschneid, 2023-02-05:
I honestly have no idea for the 48V stuff, so your conclusion is as good as mine :)

Fixes look great!

dojoe, 2023-02-06:
Alright, let's see how the board behaves once I get them back from OSHPark! Thanks again for the review!

dojoe, 2023-06-11:
Well, it took me only two years but it's done now :) The prototype works and the firmware supports the INA237 as of today's release: https://github.com/dojoe/Twonkie/releases/tag/twonkie-20230611
I merged the EPR development branch into master now - I think the hardware is fine and the firmware supports all the basic stuff.

Long term I'd like to add a few more bells and whistles but those are separate items :)

closed 2023-06-11 by dojoe