dojoe/Twonkie #5: Links to Parts

volatileConst, 2021-08-11:
Hi, I am a noob trying to solder and replicate the Twonkie.

I was wondering how to read the schematic and determine where to buy the parts apart from the wafer?

Thank you.

dojoe, 2021-08-14:
Hi! :)

The KiCad schematic file contains the vendor part numbers and DigiKey order numbers for each part designator, and I exported the full list into a .csv file too - those are commonly called "Bill of materials" or BOM for short, and this is the one for Twonkie 1.0: https://github.com/dojoe/Twonkie/blob/master/hw/v1.0/twonkie.bom.csv. You should be able to punch those part numbers into a parts supplier of your choice (I usually use DigiKey but there are plenty others) to find the parts you need. If they have parts in stock they'll happily sell you small quantities, though at an increased price. Take care to order more of most parts than you need - especially the tiny passives (resistors, capacitors...) are easy to lose during assembly.

As for reading the schematic, it probably helps to have some familiarity with USB-PD and the [original project description](https://www.chromium.org/chromium-os/twinkie), and then read up on the [device datasheets](https://github.com/dojoe/Twonkie/tree/master/ref) to understand which part does what.

Having said all that, please be warned that the Twonkie is not exactly a noob project - unless you're already familiar with hand-soldering fine pitch SMT parts you may be in for a frustrating experience. Don't let that keep you from trying, but you should know what you're signing up for :)

volatileConst, 2021-08-16:
Thank you for the thoughtful comments.

Yes, I noticed that soldering will be a challenge, especially with everything being surface-mounted. I just ordered parts from Digikey. However, due to backordering time and my school semester starting soon, I will probably not have time for this once until the semester is over.

As for closing the issue, I will close it once I start on the project itself and feel like I have no large questions left if that is good with you.

dojoe, 2023-02-04:
This issue's been lingering for a while now, so I'm going to close it. I moved my explanation into the README file so other can benefit from it as well. Once you get around to starting the project and have further questions feel free to open a new issue!

closed 2023-02-04 by dojoe