dojoe/Twonkie #8: Non-EPR hardware protection

dojoe, 2023-06-11:
Detect EPR voltages on Twinkie and Twonkie v1 hardware, force the source to turn off to protect the voltage monitor chips.

Options:
 * Sniff PD negotiation and detect EPR voltages being negotiated; inject control channel states that will abort the negotiation if so.
   * Pro: Prevents dangerous voltages completely
   * Con: Hard to implement, firmware might miss packets and thus not trigger
 * Set INA up to alert above 24V, immediately react in the interrupt handler by forcing the control channels into an invalid/reset state (possible?)
   * Pro: Easier implementation
   * Con: May be too late to protect the chips 💥

dojoe, 2023-06-11:
Prerequisite: EPR capable source and sink for testing

FCALLBOX, 2024-04-18:
hi Dojoe,
I am a little bit confused about the USB-C work when the the source are asking for 9V or 15V or 20V, the V2 HW cannot support such voltages as there is a zener 5V6 on VBUS and that the LDO MC5504 can only work at 5.5V max.

FCALLBOX, 2024-04-18:
So If I understand you does not allow USB-C negotiation over 5V ?

dojoe, 2024-04-27:
Hey @FCALLBOX!
The diode and vreg are on the analyzer logic supply which is always 5V, provided via the Micro-B receptacle by the controlling PC. The USB-C voltages are only observed via the INAxxx chips, they don't power anything on the board.
Does this help?