dojoe/Twonkie #10: Improved USB connectors

XenGi, 2023-08-27:
Ohai!

I created a 2.1 version of the twonkie with better USB-C connectors. Have a look and maybe merge it back if you like: https://gitlab.com/XenGi/Twonkie

I also added a 3D printable case which fits very nicely and uploaded it too: https://www.printables.com/model/562407-twonkie-v21-case

dojoe, 2023-08-27:
Hey, wow, that's awesome! Thanks for the change and the great case! :)

There's a problem however: Those straddle mount connectors only work for 0.8mm boards - the main reason for my weird connector setup is that it works with 1.6mm boards which are cheap and widely available. In fact the very first v0.9 of Twonkie was built for those straddle mount connectors until I realized they wouldn't work with 1.6mm boards.

Do you have a good overview over how easy it is to find a board house that will make 0.8mm 4layer boards in small quantities for cheap? That wasn't really an option back when I started the project, and I wouldn't like for Twonkie boards to be manufacturable by the likes of JLCPCB only - that would go against the original idea for me.

If pretty much any board house can make thin 4layer boards for cheap these days I'm with you that those straddle connectors are clearly better and easier to solder. If not however I'd rather stick with the current connector design.

As for the case, I'll give it a whirl and see how well it fits my Twonkies :)

Thanks again and let me know what you think!

XenGi, 2023-08-28:
I'll check with PCB fabs. I created this design kind of in a hurry to have it before CCC camp.

dnschneid, 2023-08-28:
I found that without really solid overmolding (or maybe a metal bucket-style enclosure), 0.8mm thick PCBs with a straddle mount connector made for a really weak Twinkie.  It's definitely the thing that fails first -- there's just too much torque on that mounting point.

XenGi, 2023-08-29:
So I did some research.

- [PCBWay](https://www.pcbway.com/):
  - 5 PCBs with HASL: $ 29.61
  - 5 PCBs with ENIG: $ 59.83
- [JLCPCB](https://jlcpcb.com/):
  - 5 PCBs with HASL: $ 7.00
  - 5 PCBs with ENIG: $ 17.60
- [NextPCB](https://www.nextpcb.com/)
  - 5 PCBs with HASL: $ 34.31
  - 5 PCBs with ENIG: $ 55.35
- [EuroCircuits](https://www.eurocircuits.com/)
  - 10 PCBs with HASL: € 1000.06
  - 10 PCBs with ENIG: € 1103.40
- [PCBTrain](https://www.pcbtrain.co.uk/)
  - 5 PCBs with HASL:  £ 293.95
  - 5 PCBs with ENIG: £ 323.45

I would say there are quite some options where you could get this done. I honestly don't know many other PCB fabs outside China because I'm usually not willing to pay the premium price for small quantity prototypes.
[Oshpark](https://oshpark.com/) sadly only does 2 layer 0.8mm and [Aisler](https://aisler.net/) is looking into offering 0.8mm boards in the future but at the moment you can only get them when you do a custom order for assembled boards.

hpo14, 2023-11-16:
> Ohai!
> 
> I created a 2.1 version of the twonkie with better USB-C connectors. Have a look and maybe merge it back if you like: https://gitlab.com/XenGi/Twonkie
> 
> I also added a 3D printable case which fits very nicely and uploaded it too: https://www.printables.com/model/562407-twonkie-v21-case

it would be nice if you can provide some simulate pics with the new USB-C connector,

XenGi, 2023-11-18:
Sure, Ill add some when I get to it.

dojoe, 2023-11-25:
I've given this some thought and what @dnschneid wrote makes me think the Twonkie better stay at 1.6mm. With the shell soldered to the PCB those connectors are _on_ there, no case required for stability. So while the connector mount method is certainly haxxy I prefer it over the standard 0.8mm straddle mount connectors.

Plus your cost breakdown seems to indicate that moving to 0.8mm would pretty much lock people into PCBWay or JLCPCB. I prefer leaving options open where possible and affordable four-layer boards still seem to be more widely available in 1.6mm.

I'll put a link to your repo into my readme however, so people looking for a 0.8mm variant or one with a case can easily find yours - thanks for providing it!

commit bf031ee8 (2023-11-25): Add link to XenGi's 0.8mm PCB fork

Closes #10
closed 2023-11-25 by dojoe